StevenMHernandez/ESP32-CSI-Tool #72: Bug on serial_append_time.py, does not output a csv a "binary file" with non-utf8 encoded characters

agathauy, 2022-11-05:
**Describe the bug**
Running this doesn't work. This is for active_sta by the way. I see that the last update was meant to fix this issue, but it didn't fix the issue.

`idf.py monitor -p /dev/ttyUSB1 | python ../python_utils/serial_append_time.py > my-experiment-file.csv`

Above line results to a csv file being a binary file, with the non-utf8 encoded characters also printed along with the whole serial output.

Below is an example msg. It does output CSI data to the csv turned binary file. Thought to ask you if you have any ideas for this! I feel like it may be Python version related? Not sure. I have Python 3.9.12 running.
```
$ idf.py monitor -p /dev/ttyUSB1 | python ../python_utils/serial_append_time.py > my-experiment-file.csv
/data/Espressif/esp-idf-v4.3.4/tools/idf_monitor.py:518: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if StrictVersion(serial.VERSION) < StrictVersion('3.3.0'):
--- idf_monitor on /dev/ttyUSB1 1552000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
0x4008135c: call_start_cpu1 at /data/Espressif/esp-idf-v4.3.4/components/esp_system/port/cpu_start.c:150

0x4008135c: call_start_cpu1 at /data/Espressif/esp-idf-v4.3.4/components/esp_system/port/cpu_start.c:150

```

**Tool Settings**
- Baud Rate: 1552000
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

**To Reproduce**
Steps to reproduce the behavior:
Flashed active_sta. It works properly with the serial monitor.

`idf.py monitor -p /dev/ttyUSB1 | python ../python_utils/serial_append_time.py > my-experiment-file.csv`

Results to the log being piped into the file, with the log being just the entire output + 

It's just that I can't seem to get the append time tool working properly, so thought to ask.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - ESP-IDF Version: v4.4.3
 - ESP32-CSI-Tool Git Hash: 28085b9282be446e26e60b0a34c823e927a53f8f (not sure if this is correct, but it's the latest one https://github.com/StevenMHernandez/ESP32-CSI-Tool/commit/28085b9282be446e26e60b0a34c823e927a53f8f )

**Additional context**
None

StevenMHernandez, 2022-11-05:
Thank you for noticing this new bug. I have updated the codebase (https://github.com/StevenMHernandez/ESP32-CSI-Tool/commit/478944f50af3c1f4a8e5c1e0c8ae3abc26679dee) to fix the issue. 

Please let me know if this works for you.

closed 2022-11-05 by StevenMHernandez

agathauy, 2022-11-06:
It now works thank you!

commit 478944f5 (2022-11-05): `serial_append_time.py` should not print the debug information because it is only used for writing CSI to a `.csv` file
fixes #72