StevenMHernandez/ESP32-CSI-Tool #73: How to implement "passtive" subproject to get data from AP or STA？

Sunlu2333, 2022-11-09:
**Describe the bug**
The CSI of the two ESP32 boards cannot be obtained after burning the passive subproject into the third ESP32.

**Tool Settings**
- Baud Rate: [e.g. 1552000]  (using a lower baud rate *WILL* cause issues. **this is the most common issue**)
- WiFi Channel per Device: [4]
- FreeRTOS > Tick rate (Hz): [1000]

**To Reproduce**
Steps to reproduce the behavior:
1. Build project(s)  active_ap, active_sta, passive
2. Flash project(s)  active_ap, active_sta, passive to 3 ESP32s
3. menuconfig：
[image]
[image]
[image]

5. When using channel 4 for above 3 projects, I can not sniffer any data from other device, and unable to get data from AP or STA. While CSI can be normally obtained between AP and STA.
[image]

**Desktop (please complete the following information):**
 - OS: [Ubuntu]

Sunlu2333, 2022-11-09:
Actually I have get some information when monitor "passive" and encounter some problems as follows
[image]
[image]
[image]

StevenMHernandez, 2022-11-09:
I am unable to test right now, but previously I have seen these `panic_abort` messages randomly as well. I found reflashing the ESP32 or trying a new ESP32 would often fix the issue.  Try something like this:

```
idf.py fullclean  # Cleans old build files
idf.py flash monitor
```

Sunlu2333, 2022-11-10:
> @Sunlu2333 can you please share how to get the csi data using two esp32s

Follow the readme guide and add command ”idf.py set-target esp32“ according to your type of board（e.g. esp32s2、esp32, etc）before "idf.py build".

Sunlu2333, 2022-11-10:
> I am unable to test right now, but previously I have seen these `panic_abort` messages randomly as well. I found reflashing the ESP32 or trying a new ESP32 would often fix the issue. Try something like this:
> 
> ```
> idf.py fullclean  # Cleans old build files
> idf.py flash monitor
> ```

Thank you for your answer. I found that I forgot to check the channel state information in the wifi configuration, and now it works normally. But I have a new problem: When I use one ESP32 board as AP and two ESP32 boards as STAs, which ping the same AP at the same time, and the fourth ESP32 board is assigned to passive, I can only receive CSI from one of the STAs, and the number of packets received is very small (All four boards work in channel 4). Can you give me some suggestions?

StevenMHernandez, 2022-11-10:
For future reference: `panic_abort` may be caused by not enabling:
```
Component config > Wi-Fi > WiFi CSI(Channel State Information)
```
in `menuconfig`. Thanks for sharing this!

Your second question is a good one. If you have one AP and two STAs, typically one of the STAs overwhelms the AP and the other STA suffers as a result. One solution to this is to allow time between packet transmissions so that both STAs transmit equally. Unfortunately we do not have any public implementations for successfully using multiple STAs with one AP. However, if you are interested in working on implementing and evaluating this project, please let me know. We could always use the help! Thanks!

Also, based on your screenshots, I would suggest increasing the BAUD_RATE so that more CSI data can pass through the serial port.

StevenMHernandez, 2022-11-10:
> Follow the readme guide and add command ”idf.py set-target esp32“ according to your type of board（e.g. esp32s2、esp32, etc）before "idf.py build".

@Sunlu2333 which version of the esp-idf are you using? Should `idf.py set-target esp32` should be added to the README, or is the default value always `esp32`?

Gowtham-Gorajana, 2022-11-10:
@Sunlu2333 can you please elaborate on how to print the csi data when using 2 esp32s one for active_ap and the other for active_sta

Sunlu2333, 2022-11-11:
> version: idf 4.4

I'm not sure, but without this command, I saw nothing when opening the monitor.
Before:
[image]
After:
[image]
[image]

Gowtham-Gorajana, 2022-11-13:
How to plot the collected data in the csv file  using matlab or any other tools @StevenMHernandez @Sunlu2333

StevenMHernandez, 2022-11-14:
I believe this issue has been solved. @Sunlu2333, please let me know if you have any other issues with the project.

closed 2022-11-14 by StevenMHernandez