StevenMHernandez/ESP32-CSI-Tool #78: How to adjust time stamp using SETTIME?

Arijitdutta19910601, 2023-02-07:
While collecting the CSI data collection I am unable to synchronize the ap and sta. I am running the command:

"idf.py -p /dev/ttyUSB0 monitor SETTIME:123123123123"

I ran it in ./active_ap as well as ./active_sta in both of the scenario the output is as following:

Executing action: SETTIME:123123123123
ninja: error: unknown target 'SETTIME:123123123123'
command "SETTIME:123123123123" is not known to idf.py and is not a Ninja target

Kindly suggest the exact command and path to run the command!

**Tool Settings**
- Baud Rate: 115200
- WiFi Channel per Device: 6
- FreeRTOS > Tick rate (Hz): 1000

**Desktop (please complete the following information):**
 - OS: Raspbian 32 bit
 - ESP-IDF Version: 4.3.4

StevenMHernandez, 2023-02-07:
This is how it was originally supposed to work:
Start by monitoring the serial data: `idf.py -p /dev/ttyUSB0 monitor`. Now that you have a serial connection between your computer (raspberry pi) and the ESP32, you can send data by manually typing characters and pressing enter (new-line). So just type `SETTIME:123123123123`. 

However, looking back at this, the functionality was removed back in 2019 (https://github.com/StevenMHernandez/ESP32-CSI-Tool/commit/7ac7243d2b596732c318cf9a266c8655239be58e#diff-ce62f46368cba52d2223f182e1cfe46b4366274d06598d2784efe5ac35c9a9eaR217), so it is essentially deprecated and should be removed from the documentation. However, you might still be able to add `input_loop();` to the main function of your ESP32 codebase.

**TODO: Remove SETTIME from documentation and remove any remaining code fragments.**

My suggestion is to capture the timestamp from the raspberry pi itself. Something like this: (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/serial_append_time.py)

Arijitdutta19910601, 2023-03-25:
**Thanks a lot Steven!**

closed 2023-03-25 by Arijitdutta19910601