StevenMHernandez/ESP32-CSI-Tool #79: understanding CSI data

olliec28, 2023-02-10:
Can anyone tell me what one row of the CSI data represents is it 1 row = 1 packet of data or does one row = 1 second for example. I am just trying to understand the data in order to feed it into a relevant ML algorithm. If anyone could tell me what algorithms they used as Im struggling to identify suitable ML algorithms at the moment. Thanks

StevenMHernandez, 2023-02-10:
Each row indicates 1 packet of data. So if you are receiving packets at 100Hz, you should receive 100 rows per second. Each packet contains a certain number of subcarriers which are explained in (https://github.com/StevenMHernandez/ESP32-CSI-Tool/issues/12). I hope this helps.

olliec28, 2023-02-10:
> Each row indicates 1 packet of data. So if you are receiving packets at 100Hz, you should receive 100 rows per second. Each packet contains a certain number of subcarriers which are explained in (#12). I hope this helps.

Thanks. I'm assuming in order to find the rate at which im receiving packets, that's the Tick rate In your tutorial?

StevenMHernandez, 2023-02-10:
That is a common question. I should probably update the documentation to mention this.

 It is actually in `ESP32 CSI Tool Config > ***` for the active_sta (the transmitter).

olliec28, 2023-02-10:
> That is a common question. I should probably update the documentation to mention this.
> 
> It is actually in `ESP32 CSI Tool Config > ***` for the active_sta (the transmitter).

Perfect thanks for your reply. You have any ideas on suitable mL algorithms. By reading around it looks like Recurrent Neural networks are worth attempting? Thanks

StevenMHernandez, 2023-02-10:
There are many different algorithms to try, it really depends on the task. 

In our survey (Figure 4 and Figure 1: https://www.people.vcu.edu/~ebulut/COMST22_WiFi_Sensing_Survey.pdf), we showed that signal preprocessing steps can be very important for achieving good accuracy. Then, after signal processing, you can input the model into something like an RNN or CNN.

olliec28, 2023-02-19:
Could I also ask you, how this imaginary part is calculated and if it is an estimation should the imaginary part be removed during data pre-processing so that only the real CSI data is used?

StevenMHernandez, 2023-02-20:
This script shows how I convert the real/imaginary data to amplitude/phase (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/parse_csi.py). My work so far focuses on amplitude specifically. I would recommend that you also focus on amplitude initially. Good luck!

StevenMHernandez, 2023-02-20:
Because this is not a specific bug for the codebase, I am going to "close" this issue. However, if you still have questions, feel free to leave additional comments here.

closed 2023-02-20 by StevenMHernandez