StevenMHernandez/ESP32-CSI-Tool #80: Can't modify parameters

ThomasBouiniere, 2023-02-20:
I followed all the steps required, but I can't modify the baud rate when using 'idf.py menuconfig' 
Does someone have the solution ?

StevenMHernandez, 2023-02-20:
Please share screenshots showing the issues. Does `idf.py menuconfig` say you are not allowed to change the baud rate? I need more information to understand the issue.

ThomasBouiniere, 2023-02-21:
Hello sir, thanks a lot for you answer, 
I deleted esp-idf and reinstalled it, now it works, I do have all the parameters accessible.
However, I now have issues connecting the AP and the STA. 
I have 2 ESP32, both connected to my laptop using USB.
I followed all the requirements, and put the right parameters (I think) but I don't understand why the AP and the STA are not sharing CSI Data. 

Here the parameters for the AP : 
 
[image]

ThomasBouiniere, 2023-02-21:
The parameters for the STA : 
[image]

ThomasBouiniere, 2023-02-21:
Thank you for your time sir, tell me if you need more info

ThomasBouiniere, 2023-02-21:
- configuration of the STA
cd ~/esp/esp-idf
./install.sh esp32
. $HOME/esp/esp-idf/export.sh
cd ~/Desktop/test-csi-tools/esp32-csi-tool/active_sta/
idf.py set-target esp32
idf.py build
idf.py -p /dev/cu.usbserial-0001 flash

I change the parameters using idf.py menuconfig

idf.py -p /dev/cu.usbserial-0001 flash monitor

StevenMHernandez, 2023-02-21:
You will want to make sure that the AP is set to receive CSI data (I think this is the default setting). The STA can receive CSI, but it would be easier the other way when you are learning. 

Are there any errors for the AP or the STA when you `flash monitor` them? Do you see any message that says the devices are connecting to one another?

What is `PDL-PRIVATE`? Is this an SSID of a router in your environment? If so, I think there is some confusion on the recommended standard method for using this library. The SSID should be unique to the ESP32s (the ESP32s will connect to each other).

ThomasBouiniere, 2023-03-01:
Thanks for your answer, I did not understand how to connect to the Wifi, I was trying to connect to my router. I successfully connected the STA and the AP. Unfortunately the data I receive have a weird aspect. I read the issue  #12 but I don't understand how to obtain good CSI Data. I have a lot of zeros even when removing the NULL_SUBCARRIERS as suggest. See the plots below

ThomasBouiniere, 2023-03-01:
[image]
[image]

ThomasBouiniere, 2023-03-01:
[image]
[image]

ThomasBouiniere, 2023-03-01:
I think I should obtain things like that 
[image]

PaulTbbr, 2023-03-01:
Hello, I got the same problem it seems that the phase I extract from ESP CSI is constant. I stumbled upon a research paper (https://arxiv.org/pdf/2301.00250.pdf) that seems to confirm an intended "sawtooth shape" for the original phase (as shown in the linked image and as your last figure seems to suggest).
I don't know if this is intended or if there is some parameter to tune.
[image]

StevenMHernandez, 2023-03-03:
> I have a lot of zeros even when removing the NULL_SUBCARRIERS as suggest. 

I am guessing you accidentally removed the incorrect subcarriers. Can you confirm if the subcarriers that you remove are actually all zeros?

> I think I should obtain things like that

Where did you get this "expected" data?

For phase, the ESP32 is very noisy. I have yet to dig into it (if you spend time on this, please share your findings. Community help is essential for open source project like this!)

The "sawtooth shape" is due to the phase wrapping around the plot. The plot shows phase on a cartesian graph, but the phase values are actually polar meaning they wrap around (-pi, pi). This is why "phase unwrapping" is important for visualization.