StevenMHernandez/ESP32-CSI-Tool #81: Collecting CSI data

Gowtham-Gorajana, 2023-03-22:
I can see that using idf.py monitor i can successfully print the entire CSI data in the command window but while collecting it in the csv file i could not collect the entire data instead only a csv file is being added in the folder and no data is being present .can you please help me with this.

StevenMHernandez, 2023-03-22:
Please share the exact command(s) that you are using. 

Share the successful print command AND share the unsuccessful csv command.

Gowtham-Gorajana, 2023-03-27:
[image]
this is what i am getting after opening the csv

Gowtham-Gorajana, 2023-03-27:
[image]

these are the commands i am using

Gowtham-Gorajana, 2023-04-01:
@StevenMHernandez sir please verify this and can you suggest me any errors i am struck with this

StevenMHernandez, 2023-04-01:
Try removing the 'grep' part.

Gowtham-Gorajana, 2023-04-01:
i am using it in windows and the particular command to collect the csi data in csv file is what causing me trouble

StevenMHernandez, 2023-04-01:
Check the README for the windows version of the command. Based on your previous message, you used the Linux version.

xiangshang-kok, 2023-04-04:
Sir, I would like to know the meaning of each item in the CSV file, is there any relevant introduction?

StevenMHernandez, 2023-04-04:
@xiangshang-kok This code shows how to transform the raw data from CSI into amplitude/phase (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/parse_csi.py). I hope it helps.