StevenMHernandez/ESP32-CSI-Tool #82: regarding the paper"WiFi Sensing on the Edge: Signal Processing Techniques and Challenges for Real-World Systems"

du7092, 2023-03-23:
Dear Sir, 
My name is Dheeraj, and I am a sophomore at the International Institute of Information Technology in Raipur. I am studying in electronics and communication engineering. could you kindly provide us machine learning codes so that we may use them to deploy on our dataset and change them appropriately to improve accuracy for our fall detection for older persons using Wi-Fi signals.

Arijitdutta19910601, 2023-03-25:
1. Dheeraj, it is not that difficult. Use the codes present in the Python_utils folder of this project directory. Use the codes to visualize the changes. For your particular application the machine learning algorithm and model may differ. But for standard solutions, you need to first remove the noise (smothening) from the signal, and henceforth you may chose a subcarrier from the CSI data you recieved using PCA (Prinicipal Component Analysis). After that you can label your data and apply ML algos like KNN, Random Forest, Logistic Regression, SVM etc. For your reference, look at these research works: 

1. [https://doi.org/10.3390/app122211809]
2. [https://doi.org/10.1109/ACCESS.2022.3226248]

Gowtham-Gorajana, 2023-03-27:
@Arijitdutta19910601 can you please tell on how to remove the noise of the signal if possible provide the codes also.

Gowtham-Gorajana, 2023-03-27:
@Arijitdutta19910601 @du7092 i am having trouble in collecting the data in a csv file can you please provide the detailed commaands to do that