StevenMHernandez/ESP32-CSI-Tool #83: Sir,I would like to know the meaning of each item in the CSV file, is there any relevant introduction?

xiangshang-kok, 2023-04-04:
[image]

StevenMHernandez, 2023-04-04:
See: https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/python_utils/parse_csi.py

xiangshang-kok, 2023-04-05:
What I mean by this is the meaning of these contents。I have found.
[image]
The link is https://github.com/espressif/esp-idf/blob/9d0ca60398481a44861542638cfdc1949bb6f312/components/esp_wifi/include/esp_wifi_types.h#L314

xiangshang-kok, 2023-04-05:
Thank you! SIr

StevenMHernandez, 2023-04-05:
Oh, I understand now. Typically you can ignore these items, they are for advanced use only. Unfortunately I do not have any additional documentation beyond the code that you found. 

However, I believe if you use the information in the code-comments, you can learn much more on other websites. This code is from the espressif, so I did not create the attributes, but if there is **specific** ones that are confusing, I can try to help explain some.