StevenMHernandez/ESP32-CSI-Tool #84: issue in csi data collection

dheeraj7092, 2023-04-20:
not able to run this command , cant get past this step
[image]

StevenMHernandez, 2023-04-20:
From the Readme:

> Notice, this project requires version (v4.3) of ESP-IDF.

and 

> Important: It is important that you are able to successfully build and flash the example project from the esp-idf guide onto your own esp32. If you have issues building the example project on your hardware, do not create an issue in this github repo. We will not be able to assist with general ESP32 issues (those issues that are unrelated to this project).

However, looking at the screenshot, I do not see any errors. What is the issue that you need solved? The command you run will not print any information, it should be writing data to the `.csv` file

ge419, 2023-06-22:
I have a similar issue. I'm having trouble saving CSI data into csv files -- the csv files are being created but they're empty/broken when I run the command in the screenshot below. I was expecting to see at least something being saved into the csv file since when I ran `idf.py monitor` I did see CSI data being printed in the terminal (though it was very laggy). I've also tried `idf.py monitor | findstr "CSI_DATA" > my-experiment-file.csv ` but it shows the same result. 

[image]

WGQYY, 2023-06-22:
我也遇到了这个问题

ag220502, 2023-07-15:
I am also having the same issues. I am not able to stop the CSI data into css file. But when I run `idf.py flash monitor ` I can see the following output. Can you please help me out?
[image]

[image]

mintellegence, 2023-08-23:
Hello, I'm facing issue as @ag220502 has mentioned. After flashing it shows this green and white text, but no CSI Data...

Prabal1998, 2025-04-17:
Work around solution:

In ESP-IDF 4.3 CMD, export the collected cmd output into text file and using powershell command and pipelines as mentioned below,

[image]

exported cmd text file: input.txt
output file name: output.csv
directory: same as that of input.txt file

Get-Content input.txt | findstr "CSI_DATA" | Out-File -Append -FilePath output.csv -Encoding ASCII