StevenMHernandez/ESP32-CSI-Tool #86: Does obtaining CSI information through either an active or passive station (STA) or access point (AP) occupy both cores?

jayavanth, 2023-04-21:
Wondering I can record from microphone on the second core while I'm collecting CSI

StevenMHernandez, 2023-04-22:
Interesting question. WiFi in general occupies a single core. None of the other tasks for this project specifically request the usage of the other core. I believe that you could dedicate the second core to this task. Does sound recording require 100% CPU usage on a core? 

If you already have microphone recording code, it should be simple enough to try this out by editing (https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/master/active_ap/main/main.cc).

Good luck. Please let me know if you discover any engineering issues/solutions.

jayavanth, 2023-04-22:
Thanks! I'll try it out. To elaborate on my application:

1. I have a computer that has AP that is receiving pings from ESP32-S3 for CSI
2. ESP32-S3 is also sending the CSI data it collects back to the computer
3. Computer does motion detection and responds back
4. On Core 1, mic is running and also streams to the computer and gets response back

This is seeming less likely. But I guess at least steps 1-3 should be doable right?

Not sure about mic's usage. I'm guessing it's pretty high. Will check it out