StevenMHernandez/ESP32-CSI-Tool #89: how to filter it

xiaokangkang6, 2023-05-30:
The original signal is very noisy. Could you please teach me how to filter it?

StevenMHernandez, 2023-05-30:
We discussed several denoising methods in "WiFi Sensing on the Edge: Signal Processing Techniques and Challenges for Real-World Systems" (https://www.people.vcu.edu/~ebulut/COMST22_WiFi_Sensing_Survey.pdf). We used the ESP32-CSI-Tool for all of the experiments in that paper. Good luck.

xiaokangkang6, 2023-05-31:
> We discussed several denoising methods in "WiFi Sensing on the Edge: Signal Processing Techniques and Challenges for Real-World Systems" (https://www.people.vcu.edu/~ebulut/COMST22_WiFi_Sensing_Survey.pdf). We used the ESP32-CSI-Tool for all of the experiments in that paper. Good luck.

thank you very much, i will have a try