StevenMHernandez/ESP32-CSI-Tool #96: documentation for the excel sheet

wael-ejkhidem, 2023-09-03:
hello Mr. Hernandez, 
Thank you for the helpful material, i just have some issues. 
first issue, sometimes when i run the code, i got a blank excel sheet.
second issue, could you please provide me the documentation of the excel sheet, because some parameters and values are unknown and undefined.
third issue, i tried python code for plotting the CSI data, but unfortunately its not working. 
thank you, 
Wael.

wael-ejkhidem, 2023-09-05:
@StevenMHernandez

wael-ejkhidem, 2023-09-05:
" 
UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.show() 
" 
this what i got, when i try to plot the csi signal using the provided code

Junye97, 2024-01-25:
Hi @wael-ejkhidem, to my understanding: 

1. If there spreadsheet is empty, I think the device is NOT receiving any CSI. Can you see CSI output printed on Serial Monior? 
2. Can you please specify which parameters are not clear? If you are asking for the CSI data format, it seems to be following [imag real imag real imag real...]:
https://github.com/StevenMHernandez/ESP32-CSI-Tool/blob/cce5b1a055fae5b01b30d1d39c5cfa1e0c656696/python_utils/parse_csi.py#L14