StevenMHernandez/ESP32-CSI-Tool #101: Multi-antenna support

mweisel3, 2023-12-07:
I was wondering if this tool would support collecting CSI from an antenna array. It seems like ESP32 supports multiple antennas through a switch (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-multiple-antennas), would the tool pick up the CSI from all the antennas, or only the selected one?

shuizhamouling, 2024-05-06:
I am also facing the same issue. Has it been resolved?

mweisel3, 2024-06-06:
As far as I can tell, this isn't an issue with the tool, but rather with ESP32. I don't think it exposes the CSI related API when using an antenna switch (I assume that data isn't available). I could be wrong, but that's what I gathered. 
Presumably, this tool would more or less work if an ESP32 chip that truly supported multi-antenna CSI data was available.