StevenMHernandez/ESP32-CSI-Tool #104: Figure 1 not open

Cizzy5, 2024-03-31:
When I run Python csi live.py with esp idf, Python not respinding.
[image]

StevenMHernandez, 2024-04-06:
@Cizzy5, do you see any CSI printed with the standard `idf.py monitor`?

@AsciiPro99, any suggestions for why there is not any CSI printed?

AsciiPro99, 2024-04-06:
If you can print CSI values to monitor fine with the above command. It is likely due to missing imports. Confirm Stevens inquiry and then post your package list.

Cizzy5, 2024-04-15:
> there is not any CSI plotted?

Yes i see CSİ PRİNTED with standard idf.py monitor, i also visualized csi
data by Matlab, but when I run live Python code not open matplotlib figure.
Thanks for you interested.

AsciiPro99, 2024-04-15:
Could you post your package list?